Gameboypi/SPW #2: Eagle Files and button placement for X Y buttons?

freeacount1, 2018-10-24:
Hi, is there any chance you can upload the Eagle files? I would like to edit the pcb for DPI LCD output if possible thank you.

woozer, 2018-10-25:
I would add a charger (booster) protection circuit and onboard I2S audio, although this solution is perfect for people less experienced with soldering!

DPI output would only be possible if you plan to use e.g. a Atmega32U4 for button input, and are OK to lose I2S sound, unless you add some USB audio (CM102 + FE1.1s).

Button presses could also be handled by an I2C IO expander connected to the cam port, with a device overlay like done on https://othermod.com/ 

W

freeacount1, 2018-10-26:
I don't mind if I lose the I2S sound, I was going to add a USB hub along with a USB soundcard/pam8302 and MCP23017 for I2C controls.

as for the battery and boost feature i was gonna use GreatScott's  circuit

https://www.youtube.com/watch?v=Fj0XuYiE7HU

https://easyeda.com/GreatScott/LiPoChargeProtectBoost_copy-3d9f4c3ddc7347d7861776340b9b90b7

woozer, 2018-10-26:
> If you could provide the eagle files for the SP board you designed I would like to give it a try.

That's not up to me! ;)

BTW, the cm102s+ USB audio chip is supposed to support up to 3 watts to a direct speaker connection, so you won't really need an amplifier.

W

Gameboypi, 2019-01-09:
Sorry guys I can’t share my Eagle board files! :(

closed 2019-01-09 by Gameboypi