Gameboypi/SPW #5: Questions about Build

zarderxio, 2019-01-05:
Hi All, 2 questions: 

For the ADS1015 optional battery monitor, how does that get implemented?

The link port is tad expensive online, does it have a functional purpose besides filling space? 

Thanks!

Gameboypi, 2019-01-09:
There are five pads behind the battery terminal for the ADS1015, maybe did not explain it yet.. But these links will help you.
https://sudomod.com/forum/viewtopic.php?f=42&t=4585&start=140

Battery monitor code
https://github.com/HoolyHoo/Mintybatterymonitor/blob/master/README.md

I use the link port with little modifications
as micro usb input you can ignore it if you want, it's optional

closed 2019-01-09 by Gameboypi

Gameboypi, 2019-01-30:
That’s how to wiring ads1015 
Sorry for bad image i have no access right now to pc “add 10k ohm resistor to a0 “

https://imgur.com/gallery/Y3OOpzn

Unfortunately there’s no battery monitoring code yet for retropie 4.4...