Gameboypi/SPW #14: GPIO 13 - M button unassigned

BRaSSaS, 2019-03-24:
After going crazy, and reviewing the key assignments, I have seen that the key that is marked "M" in the circuit, which corresponds to the brightness key of the SP, is not assigned and can not be used.
I have reviewed the configuration file in /home/pi/Adafruit-Retrogame/retrogame.c
And all GPIOs are assigned except for GPIO 13, which corresponds to the M key.
Am I confused or am I doing something wrong?

Gameboypi, 2019-03-25:
Nothing is wrong, yes it is not used, its function was as a safe shutdown button but I have problems with the code so currently is not used and you can register the button and use it as you want

BRaSSaS, 2019-03-25:
I have left it configured as:
{13, KEY_LEFTALT}, // M button
Thus it can be used for hotkeys in all emulators. There are some like the mame4all that in those that do not work retroarch hotkeys, and in which you can not recreate combinations of keys if one of them is not a modifier (Ctrl, Shift, Alt)
So in this way I leave all the hotkeys configured with that key

phillymadison, 2019-07-15:
So currently you can't config the M button in the config menu but have to change the code for it to work? I've been trying to assign it has one of the controller buttons but haven't been successful using the config controller option. Wasn't sure if it was my sodering or bad buttons.

bangooh, 2020-03-19:
> I have left it configured as:
> {13, KEY_LEFTALT}, // M button
> Thus it can be used for hotkeys in all emulators. There are some like the mame4all that in those that do not work retroarch hotkeys, and [...]

Can I map the 'm' button by adding the line you mentioned in the /boot/retrogame.c file?

Eilermoon, 2023-04-20:
> So currently you can't config the M button in the config menu but have to change the code for it to work? I've been trying to assign it has one of the controller buttons but haven't been successful using the config controller option. Wasn't sure if it was my sodering or bad buttons.

I'm actually having this same problem now. Should this be button be able to be assigned? Currently if I'm in RetroArch and try to assign it to something, nothing happens. Is there something I'm missing?