Gameboypi/SPW #15: Ground/battery -

TobyDude, 2019-04-04:
Hi,
I had a small question? How is the pi connected to ground, I can't seem to find a point to which
the pi is connected. 

Thanks in advance.
Toby

BRaSSaS, 2019-04-04:
It's connected trow pin 6, 9, 25 and 39.
If you look closely at the raspberry pinout scheme, and the spw circuit you will see that all the rpi ground are being used

TobyDude, 2019-04-04:
Thanks for the quick response, But where does the battery – go to on the pi or does that connect to one of the ground pins?

BRaSSaS, 2019-04-04:
I do not understand the question, the battery is connected to the battery connector/termina in case you use an original gameboy battery, or if you use another 3.7v generic battery, you will have to connect it to the TP4056 CHARGER (B+/-) or reuse the battery connector pads.

TobyDude, 2019-04-04:
So then the pi receives power from out+ and out- on the tp4056. But where does out- connect to the pi?

BRaSSaS, 2019-04-04:
Same answer, to the pin 6, 9, 25, 30, 34 and 39. It's a ground, it's general ground.

TobyDude, 2019-04-05:
Alright thanks man. 😃

closed 2019-04-28 by Gameboypi