Gameboypi/SPW #16: MicroUSB ports

hpagseddy, 2019-04-22:
Hello, i want to use Raspberry Pi’s HDMI and microUSB ports. But they are too far from the side of the Gameboy. Should i just drill a hole or extend them with wires? Thanks.

Gameboypi, 2019-04-26:
Ummm, I don’t know if it will work for you, but extend them better than drill a hole

BRaSSaS, 2019-04-26:
I use the usb / otg without problems, and the micro-hdmi adapter just had to file a bit. It is true that the micr-usb of charge, is covered, but it is not something that you are going to use.
Anyway, I agree that the rpi is poorly positioned, should be more on the edge, to be more comfortable, and cover it with a modified cartridge, or something like that. But mainly for the SD card, it is very badly positioned, it could be on one side and with a small modification of the carcass that was possible to remove it and put it in, or at least that the power switch was not in the middle.

hpagseddy, 2019-04-27:
Also, i am curious about which side the ports are facing. Are those towards to cartidge slot or the back side of the case

Gameboypi, 2019-04-28:
Hdmi and micro usb are facing cartidge slot said

closed 2019-04-28 by Gameboypi