Gameboypi/SPW #17: PCB Gerber file type

Joe-Alfaro, 2019-04-25:
Do you know of a printing service that will print the PCB files as is? The companies I tried all require RS-274x file types. I have never printed PCBs before so I'm not even sure what that is or if it is possible to convert the original files to that type.

BRaSSaS, 2019-04-26:
There are a lot of services that work with gerber file type.  In fact, it is the most used by this type of pcb printing companies. Many even have an online viewer so you can see how the circuit is before ordering it.
I work with jlcpcb or pcbway

Joe-Alfaro, 2019-04-27:
Hi, I checked probably five companies and everyone failed the audit because they said they needed the RS-274x file type, including pcbway. I did mean to update this issue though because I came across jlcpcb only because they were sponsoring a video on youtube and everything went perfectly and for the cheapest price I had come across. Thank you for your response, this is the first time I have ever tried to print a PCB before and hopefully someone else will see this thread and just go straight to jlcpcb instead of wasting a bunch of time. And thanks gameboypi, I'm very excited to go after this project, your instructions are really clear and it's so awesome of you to put this stuff out there for us!

closed 2019-04-27 by Joe-Alfaro