Gameboypi/SPW #18: Hi,

AlexNT1, 2019-04-26:


AlexNT1, 2019-04-26:
Hi,
I have question.How are pi and PCB connected?
It looks like soldering the pi's GPIO part and pcb.
Is this all right in this way of thinking.

Thank you for reading this question.and I'm glad to answer this question.

BRaSSaS, 2019-04-26:
Yes, everything is designed to be welded on the main PCB, it is designed so that the welding holes / pads coincide on the 2 plates. This is true for everything, for the PRI board, the sound board and the battery charging plate.
Anyway, you can see the assembly process on this git, it's very well illustrated.

Gameboypi, 2019-04-26:
Thank you BRaSSaS! :D

AlexNT1, 2019-04-26:
Thank you for answering　BRaSSaS!
and thank you for leading this question Gameboypi!

It is my first time to do electronic work, but I will challenge while enjoying it

Gameboypi, 2019-04-28:
Good luck dude!

closed 2019-04-28 by Gameboypi