Gameboypi/SPW #24: White Screen and Connection Help

phillymadison, 2019-07-14:
Hi all I bought the materials to make a few of these so I have 2 being made right now. Both are giving me a white screen on power up. I Switched the screens to the other main boards to see if I sodered something wrong, flashed different sd cards ( I got the image from here and they work if I connect them to a different pi on a tv) and checked if I had any shorts or if something wasn't sodered correctly with a multimeter. I think it might be my connection from the screens ribbon cable to the mini board. Is there a method to connect the screen to the mini board? Just a little sodered, and patience? Any help would be appreciated. Thank you

Gameboypi, 2019-07-14:
Check out this to troubleshoot the issue 
https://github.com/Gameboypi/SPW/blob/master/Screen%20troubleshooting/README

use some flux and reflow the screen pins

phillymadison, 2019-07-15:
Ahhh the extra flux was a huge help. I have 2 fully working screens now  thanks

closed 2019-07-15 by phillymadison