Gameboypi/SPW #25: Backlight flickering (when resistor is used)

moon002, 2019-07-16:
Hellooo!
So I **had** an issue for a while with the backlight flickering on my SPW - I wired everything according to the instructions, however on two separate (new) 3.2" LCD displays the backlight would flicker quite a lot, especially during startup and it would come and go during usage. These displays were the ones linked in the wiki. Interestingly it never happened on a third lcd (w/ touch panel). Also of note is that the displays would not flicker "out of the box", only after 5-10 minutes of initial usage would they begin to flicker (enough time to close the casing and sit on the couch..).

So after many agonizing hours of troubleshooting, I was able to stop the flickering by simply bypassing the 3.9R resistor for the backlight power.. I had tried various other resistor values - it took a while to brave skipping the resistor altogether! I know it's not ideal to skip this resistor, but could anyone shed light on why this is happening? And I'm thinking it's reasonably safe to continue without a resistor, it has been fine so far, but maybe I'm wrong? I wonder if there's a resistor built into the display..

I have been theorizing that it could be:
- Sub-standard resistors (eBay purchase)
- Sub-standard LCD displays
- Defective Raspberry Pi
- Electromagnetic interference somewhere
- A resistor is not needed

I'd appreciate any input!

N.B. Avoid connecting GND and battery GND in any manner of troubleshooting, it will fry your LCD.

Gameboypi, 2019-07-16:
Umm that’s weird 
Are using right resistor value?
Does the battery have enough power? Because that’s happened when battery runs out.
There’s no problem if you paypass this resistor it’s just for dimming the backlight.

moon002, 2019-07-16:
Hi @Gameboypi 
Thanks for your great work on this project!

- Yes I was using a "3.9 OHM 1% 1/8W 0805" resistor as specified. The flickering got more intense as I put a two or three in series, i.e. 7.8 / 11.7 ohm, and of course the backlight was generally darker.
- Yes the battery was fully charged, but regardless it was happening while running just from direct USB power to the RPi.

If this has not happened to anyone else, then I can only assume it's a tolerance issue in one of my components..

Do you think GPIO PWM -> transistor would be a good way to control the backlight brightness, with view to eventually having on-screen controls?