Gameboypi/SPW #30: RPI wont boot

brueckspecht, 2019-09-15:
First of all thank you for this awesome project and the excellent documentation.
I have a weird issue: After soldering the parts to the PCB I tested the build with a power adapter in the PI: works as expected, Retropie boots. Afterwards I plugged the power adapter in the TP4056, turned the switch on but the RPI wont boot. So I checked the voltage, the PI gets 3.7 to 3.8 V which should be enough I think. As a result I built a second setup (only a PI, TP4056 and the switch) with the same result.
Do you have any idea what might be the reason?
Thanks in advance.

phillymadison, 2019-09-16:
Do you have the battery installed too? Pretty sure you need the battery to complete the circuit

brueckspecht, 2019-09-16:
@phillymadison Thanks a lot that fixed it ;)

closed 2019-09-16 by brueckspecht