Gameboypi/SPW #31: white screen...

brueckspecht, 2019-09-18:
Hello again,

today I have another issue:
I soldererd everything together now but the screen stays white when powering up.
It is a 2.8 ili9341 screen (as described in the wiki part list)
Things I tried to solve the problem:
1. I followed the screen troubleshooting tutorial --> everything ok
2. I double checked my soldering and resoldered with a lot of flux --> Multimeter says everything ok
3. I even multimetered from the GPIO´s directly to the screen --> everything ok
4. I forced the screen to boot in 320x240 resolution in the config.txt (which is not necessary at all because the image is compatible with the ili screens)
5. I reburned the image to the SD card
6. I put the SD card into the slot of another GB project --> works!

So do you have any more ideas?
Maybe the screen is broken?

Thanks in advance...

phillymadison, 2019-09-18:
Are you giving it enough time to boot up? The screen starts white for 10-15 seconds then black then the retropie splash screen will load up

brueckspecht, 2019-09-20:
Hello @phillymadison and thanks for the reply.

Yes I give it enough time. I´ve been waiting for several minutes.
Today I tried to start the screen on my second setup with no result as well.
Afterwards I switched the little screen PCB, soldered the screen to it and booted the device again.
The screen stays white. A little pressing at the edge of the screen makes it flickereing so I guess the screen is broken.
I already ordered a new one but it will take some time until I will get it.
I will post an update when the new screen arrives.

phillymadison, 2019-09-21:
I have my own box of broken parts that taught me little lessons along the way.

Is the orange ribbon cable coming from the screen to the small board touching the metal on the back of the screen at all? I had this issue where I would glue the small board to the back of the screen and the small components on the orange ribbon cable would tough the metal on the back of the screen and make the screen go white. A strip of electrical tape fixed that. 

Lastly are you sure you didn't soder one of the screen connections to another or not well enough? That 18 pin screen isn't the easiest to soder

Frenzis78, 2019-09-22:
I had the same issue and all test were ok...but in the end I had to resolder the cable connectors with a lot of flux and solder wire... After that all went fine.

brueckspecht, 2019-09-28:
Hello everyone and thanks for the answers.
Today I gave it another try and soldered the connectors and the screen once more, unfortunatly with the same result.
Hopefully my order will arrive soon...

brueckspecht, 2019-10-09:
A few attempts later the screen works.
This is witchcraft!!! ^^

closed 2019-10-09 by brueckspecht