Gameboypi/SPW #33: Button problem...

brueckspecht, 2019-10-27:
Well hello again,
today I have a new issue which I hope you can help me with.

The "up" button doesn´t work...
Now let me explain what I tried so far and why I open an issue for this:

1. GPIO Test with the multimeter as described in the buttons troubleshooting section - Check
2. Button functionality test with the multimeter - Check
3. 1 + 2 or: Multimeter the button and the GPIO simultaniosly, press the button (beeep) - Check
4. Retropie Config: Button reconfig won´t recognize the "up" button at all - Check

So it seems that everything is set up correctly but the hell of a button simply doesn´t work as it should.
Do you have any ideas?
Thanks in advance.

Gameboypi, 2019-10-28:
That’s weird, Could you please check if there’s any shorts between “up” and 3.3v or Left trigger button?
If there’s nothing try to reflash the retropie img & reflow up gpio

brueckspecht, 2019-10-29:
Hello Gameboypi, thank you for your reply.
I checked the GPIO´s for shorts as described, reflashed the image and resoldered the GPIO.
No luck here.
I will see if I can test the functionality of the GPIO22 itself somehow...

closed 2020-05-22 by Gameboypi