Gameboypi/SPW #34: Sound Issues

Sinnohman, 2019-12-23:
I can't get the sound to work. I've confirmed everything is soldered correctly, so I think there's some issues with my retropie sound settings.
Can anyone please confirm the Sound Settings from within retropie.

Gameboypi, 2019-12-23:
Thats the link how to setup the i2s audio 
https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/raspberry-pi-usage

or you just re-install the retropie image from my github it's pre installed with all settings.

Sinnohman, 2019-12-24:
I've reinstalled the image and have sound now, but it's really distorted. Also, the area where the speaker is gets really hot, I'm assuming that's not normal?

Gameboypi, 2019-12-24:
Yes that's happens when you have a bad soldering & short between the Pi and i2s module
re check your soldering.

closed 2020-05-22 by Gameboypi