Gameboypi/SPW #35: not an issue, more a help request

Frenzis78, 2019-12-25:
Hi, 
I build a second unit, just beautiful.
I would like to install the retropie-manager, but I do not have a keyboard to enter my github credentials,  I wonder if I could do it over SSH, but I'm a real noob. (wifi was imported via the boot txt file)

Any idea?

Thanks

Gameboypi, 2019-12-25:
No need for external keyboard, yes you could do it over ssh with “cyberduck”
Go -> Open in Terminal

Frenzis78, 2019-12-25:
Hi, I found a ay via ssh usin putty... just found out that the old retropie-manager intalled is not available anymore, I now need to find a way to change it. thanks anyway

Frenzis78, 2019-12-25:
I managed to do everything by SSH. Great easy tool. Just a note for your great project, the retropie-manager is obsolete (the link in the version I have downloaded) you might want to signal that an update is mandatory to have this experimental package installed.

Anyway great project, thanks Gameboypi

closed 2019-12-27 by Frenzis78