Gameboypi/SPW #36: Screen Issue

ThRealLorenz, 2019-12-30:
Good morning y'all,
I've come across a very strange problem, 'cause my screen doesn't work but I've double checked all the contacts and the continuity beeps. I state that I haven't finished all the soldering, I've only tinned the buttons and the screen and I'm powering it with the microUSB port on the Raspberry

phillymadison, 2019-12-30:
Is the screen white when you turn it in or is it just nothing?

ThRealLorenz, 2019-12-30:
It turns white when I turn it on

phillymadison, 2019-12-30:
Ok that’s good the screens getting power then. Let it sit for a little bit after you turn it on (30 seconds to a minute) when it’s working correctly it starts up to a white screen then clicks over to retro pi

ThRealLorenz, 2019-12-30:
After 5 minutes of waiting nothing happens :( but I've just found out that gpio pins 23 and 26 are shorted together. Could this be the cause?

phillymadison, 2019-12-30:
If it’s solder spot 23 that’s shorted than probably yes. The gpio’s have their own numbers that aren’t in order but if it’s pin 23 counting from the square pin and going right and down than yes cause that’s one of the troubleshooting pins that should be check when trying to fix a bad screen. Even if it isn’t shorts are never good so fixing it now will save time later

ThRealLorenz, 2019-12-31:
I've re-soldered every component and now it works! Thank you anyway @phillymadison

closed 2019-12-31 by ThRealLorenz