Gameboypi/SPW #37: Can't find a power switch

ThRealLorenz, 2019-12-31:
I've ordered a wrong power switch but now I can't find any power switch on ebay and the ones i've found on aliexpress haven't clear descriptions and I don't want to buy an another wrong switch. Can someone link me the right switch?

NO-ob, 2019-12-31:
https://www.ebay.co.uk/itm/Power-On-Off-Switch-replacement-Original-Nintendo-GBA-SP-Game-Boy-Advance-SP/143407750866?hash=item2163c49ed2:g:YBEAAOSwyfFbcdPr

TheRealLorenz, 2019-12-31:
@NO-ob It doesn't ship to my house :(

phillymadison, 2019-12-31:
These are stupid expensive now but here’s a US seller 
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F193259460931

I’ve gotten these to work which are a lot cheaper Just a little harder to install but they do work 
https://a.aliexpress.com/dpg9Kmo0

TheRealLorenz, 2019-12-31:
Are the ones from aliexpress that you linked me compatible with the piece of plastic of the switch?

phillymadison, 2019-12-31:
Yes it needs to be moved up a little cause it’s a little smaller than the real one but once it’s in and all closed up you can’t tell the difference

Frenzis78, 2019-12-31:
I bought this one
https://a.aliexpress.com/eZ3wh2w8

bademux, 2020-03-01:
Psp1000 matches, but it is momentary switch

closed 2020-05-22 by Gameboypi

mousemat86, 2020-06-14:
I cannot find any SP switches anywhere. has any one found a successful alternative from aliex? thanks