Gameboypi/SPW #40: Is there possible to add a headphone output inside?

MestreALMO, 2020-03-08:
I never messed around with eletronics so i don't know if this is something hard or simple, i really want to use my headphones with it if possible. How can I make this?

[image]

phillymadison, 2020-03-08:
I’m pretty sure it’s 100% possible. The easiest way would be to use Bluetooth head phones. If not try a micro usb head phone adapter and plug it into the micro usb hub from the pi. I’m not sure if there’s a way to solder a jack in there and have room. I’m sure someone else could comment about that.

Gameboypi, 2020-03-11:
Technically yes, by cutting off the speaker traces and turning the traces to the headphone jack (with sense pin) and run wires from the headphone jack to the speaker because the i2s chip doesn't have a HS pin, so you need to do some wiring and im not sure if there is any free space for that.

closed 2020-05-22 by Gameboypi