Gameboypi/SPW #42: Upgraded display driver

ssjowowo, 2020-03-29:
I found that rpi-fbcp is having some performance issues, so I replaced the one bundled in your image with the fbcp-ili9341 one found here: https://github.com/juj/fbcp-ili9341

The difference is night and day as it provides very smooth transition between frames and removes that choppy framerate in rpi-fbcp when playing games.

If anyone is interested read below.

You can follow the readme for build instructions. The cmake command that gave me the best results is the one below:

`cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=0 ..`

Once you compile, just replace this line in /etc/rc.local with the binary out file:

```
#/usr/local/bin/fbcp &
/home/pi/fbcp-ili9341/build/fbcp-ili9341 &
```
NOTE: I also have an overclock of 800mhz which helps things a bit

Gameboypi, 2020-03-31:
I will take a look at it Tyvm!!

pep3175, 2020-04-18:
> I found that rpi-fbcp is having some performance issues, so I replaced the one bundled in your image with the fbcp-ili9341 one found here: https://github.com/juj/fbcp-ili9341
>
> The difference is [...]

Hi, very interesting ! Are you able to run gba games ?

ssjowowo, 2020-04-18:
Yes gba runs ok. Tested almost all emulators that rpi 0 can run and found no issues

pep3175, 2020-04-18:
@ssjowowo  Great. I'll give it a try. That is my main issue with this build. Thanks for your upgrade.

gambino37, 2020-04-21:
Wow,  the juj driver is a night and day difference!

mousemat86, 2020-05-27:
any chance this update might be included in a new version of the SPW retropie image? @Gameboypi

mousemat86, 2020-06-19:
adding this for reference

```
sudo apt-get install cmake
cd ~
git clone https://github.com/juj/fbcp-ili9341.git
cd fbcp-ili9341
mkdir build
cd build
cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=0 ..
make -j
sudo ./fbcp-ili9341
```

stanyertom, 2020-07-26:
Is this installed onto the image now or do I have to run these commands?

Eilermoon, 2021-06-30:
So I don't have a keyboard or way to input commands into my Pi Zero.  Is there a way to do this to the SD card on another PC? I didn't plan for the lack of USB ports haha...

Eilermoon, 2023-05-24:
Hey guys, hopefully someone has some advice for this. I did a fresh new install of RetroPi and went through this screen driver process. I can get the screen to work going through these commands, but my rc.local file isn't working on boot, the screen stays white. Any ideas why this would be? I added ```/home/pi/fbcp-ili9341/build/fbcp-ili9341 &``` and commented out the ```#/usr/local/bin/fbcp &``` line in my rc.local file.

ssjowowo, 2023-05-25:
can you please paste your rc.local contents? if you run `sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341` do you get the display to work?

Eilermoon, 2023-05-26:
I've actually got it going now after doing a fresh install! I cleared my overclocking settings in my /boot/config.txt file, that must've been too much? 

I'm currently fighting with it constantly restarting though, not sure why. It will boot into emulatorstation, then the screen will go white after a few minutes and boot up again

Eilermoon, 2023-05-26:
> can you please paste your rc.local contents? if you run `sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341` do you get the display to work?

To answer your question though, it looks like I screwed up the install somehow. Not sure how, but a fresh install fixed it. and after removing the overclocking, the screen was legible again - it was previously all smudged?

Eilermoon, 2023-05-26:
I also would love to add the ```-DLOW_BATTERY_PIN``` command to show a battery low icon when it detects a low voltage on a pin. I'm note sure what pin to use though with this build, anyone have any ideas?

ssjowowo, 2023-05-28:
Glad you got it working. Overclocking usually causes issues with the fbcp-ili9341 driver (https://github.com/juj/fbcp-ili9341#the-display-works-for-some-seconds-or-minutes-but-then-turns-all-white-or-black-or-freezes).

As for a spare GPIO, if you are not using the I2C interface, you can use the SDA and SCL test pads on the pcb. These are mapped to GPIO2 and GPIO3 respectively 

[image]