Gameboypi/SPW #43: Screen Trouble

Morpewpew, 2020-04-17:
Hello everyone,
So the other day I started to assemble everything together, and when I got the to the step in which to test the screen it did not turn on. I re soldered everything and still no luck. I then followed the the troubleshooting guide, in which I had some continuity error between the 3.3V pad on the mini pcb board and the screen connector. After I fixed this and it once again had continuity but the screen still did not turn.  The while testing the continuity I noticed that if I touched the terminals of the multi-meter  to the 3.3V and miso pads it would actually turn on but only then and the screen would be blank. The pi seems to to work fine, and so does the screen. At this point I am at a loss and if y'all have any suggestions as to were to start next I would appreciate it. 
Thanks

Kalamangus, 2020-04-19:
You probbably want to also check, if any pins are shorted together. 
Try the hdmi functionality on the pi too (on my pi with the included retropie image it says, that the pi gives out an unsupported signal, which should be normal, because of the config file, that comes with the image). 
I f'd up 2screens so far, because i bent the ribbon cable too much back and forth (i got the backlight showing and if i moved the ribbon cable the backlight went out). 
Try to also check from the pad from your raspberry pi to the actual screen ribbon cable and check any shorts over at the pi side (any pins that are not used "should be fine"). If you do find a broken trace you can always use some kind of wire like wire-wrapping-wire, that is a very big gauge (bigger gauge = smaller wire) and is awesome for this stuff. 
If you don't get any activity on the screen like a backlight or if the backlight flickers you might be having power problems, if so just plug it in to the wall with a usb plug and see, if it works any better.

Morpewpew, 2020-04-21:
Thank you for the advice. I tried a lot of your suggestions, and I got the screen to start to flicker when being turned on. Your suggestion to try the continuity from the pi to the screen ribbon cable has helped a ton. For some reason the continuity doesn't line up with the images shown from the troubleshooting. For example the miso (GPIO9) currently has continuity with pin 9 of the ribbon cable, and it is like this for the other connections. So my guess is that it may have something to do with the connection from the general pcb board to the smaller one that connects to the screen. When ordering parts I believe I got the reverse side ribbon cable instead of the same side. Could this be causing my issues?

gambino37, 2020-04-21:
Absolutely.  You have the wrong ribbon, basically your ribbon "pins" or "paths" however you want to look at it are reversed.

Morpewpew, 2020-05-05:
Okay so I order another cable and finally got around to testing it but it had the same problem. This is the cable in question https://www.molex.com/molex/products/part-detail/cable/0152670234, did I make the same mistake again and not get the right cable. I now noticed it was type D and the cable from the Ali Express is type B if I remember correctly. So am I just being dumb and ordering the wrong thing again?

Kalamangus, 2020-05-05:
Over here on Wikipedia https://en.m.wikipedia.org/wiki/Flexible_flat_cable
it says, that flat cable type d has the pins faced opposite on each ends.
The one you want has the pins on both sides facing the same way so it
should be type A according to Wikipedia.

Morpewpew, 2020-05-05:
Yeah I have come to realize I was just being dumb, thanks for all the help everyone.

closed 2020-05-05 by Morpewpew