Gameboypi/SPW #46: GBA SP power switches

mousemat86, 2020-06-14:
I cannot find any SP switches anywhere. has any one found a successful alternative from aliex? thanks

phillymadison, 2020-06-14:
Hi Mousemat I ran into this same problem. It seems there’s a shortage. I’ve gotten a PSP switch to work. It’s not a perfect match takes a little soldering skill but it does work. Here’s a link to the ones I bought.

https://a.aliexpress.com/_Bfgeo37C

mousemat86, 2020-06-14:
hi thanks for getting back to me @phillymadison 
that's interesting -  I actually already tried one of these, whilst it worked when I tested it once I put the case together the plastic button didn't slide far enough to turn on the board. 
did you modify the switch or positioning at all? I assume because it has a spring it's a momentary switch. 
thanks!

phillymadison, 2020-06-14:
If I remember correctly I had to solder it a little more up on the board. You’re right about it being a spring switch but the plastic switch holds it in the on position. It was definitely a trial and error install. Where having an actual GBA SP is a smooth installation. 

You could also buy a broken GBA SP (whole thing or just the board which ever is cheaper) and use the switch from it. Can take the L and R buttons too. I was having issues finding authentic 4 pin ones for a few of my builds at the end.

mousemat86, 2020-06-21:
thanks @phillymadison 
I got the PSP 1000 switch working - like you say some creative soldering seemed to help the positioning

stanyertom, 2020-07-26:
I managed to get an Nintendo ds lite switch to work if that helps anyone