Gameboypi/SPW #47: Screen brightness problem

mauriesvh, 2020-07-08:
First of all, thank you very much for the build instruction provided. 

I am finishing building my GBA Sp, however, I am having some issues with screen brightness. As can be seen in the pictures, the brightness is pretty poor. It is hard to see anything on the screen in normal daylight. I can not figure out what is causing this issue. I have the small screen as in the parts list. Battery or power supply does not give a difference. Maybe something to do with the screen resistor? For that I choose a 3.9ohm one. Hope you can help me!

[image]
[image]

Kalamangus, 2020-07-15:
Do you see the backlight turning on?

mauriesvh, 2020-08-17:
Yes I think I do. After turning it on, the screen goes from black to slightly lit. However, it is quite hard to see in normal daylight.

Kalamangus, 2020-08-17:
Have you followed the screen troubleshooting guide?

mauriesvh, 2020-08-17:
Yes I did. Just double checked the soldering and all connections are fine

Kalamangus, 2020-08-17:
The screen should turn white and then to black and after around 5-15 seconds it should show the splash screen.

mauriesvh, 2020-08-17:
Yes it does that, but the brightness is so low you can hardly tell. In the pictures above the room was quite dark and only then you can see a bit what is happening on the screen.

Kalamangus, 2020-08-17:
You might have accidentally damaged the screen ribbon cable. Is the screen getting enough power? And can you check the other ribbon cable (the one that goes thru the hinge) by checking it from the screen connector to the main board's connector?

mauriesvh, 2020-08-17:
The ribbon that goes thru the hinge is fine, just checked. Also, I powered the device using battery and by using USB both gave similar results power-wise. 
The screen ribbon connections are fine, however I can be damaged but I doubt that.
I was thinking power-loss can occur when choosing the wrong resistor for the screen PCB. I choose a similar one as in the parts list, so I don't know if that can be the problem.

Kalamangus, 2020-08-17:
You can try to bypass the resistor and see if anything changes.

mauriesvh, 2020-08-18:
Thanks for the help, I think I fixed it. When pushing a bit on the screen on specific spots, somehow the brightness increased. Difference is night and day now. Screen might be a bit broken.