Gameboypi/SPW #48: Not issue but addition to guide if allowed: Safe shutdown Via brightness button

stanyertom, 2020-08-02:
https://core-electronics.com.au/tutorials/how-to-make-a-safe-shutdown-button-for-raspberry-pi.html

I used this guide 

But first you will have to install gpiozero by running 

pi@raspberrypi:~$ sudo apt install python-gpiozero

In the command line 

Then I followed the guide here but changed the line;
“ stopButton = Button(26) # defines the button as an object and chooses GPIO 26”
To;
“ stopButton = Button(13) # defines the button as an object and chooses GPIO 26”

I also changed the time to (3) so you hold the button for 3 seconds to shutdown.

mousemat86, 2020-08-03:
nice! thanks for sharing

01Turgeo, 2020-09-22:
Thanks for sharing! Looking forward to my build, but not a huge software guy and have to ask: can this button be used for adjusting brightness as it was originally intended?