Gameboypi/SPW #54: PCB does not meet JLC PCB requirements - Break away tab is too small

adecker, 2021-04-18:
JLC will not manufacturer this PCB anymore due to the following requirement

[image]

Are the source PCB files available anywhere or could you update the break away tab to meet this minimum? 

Thank you

WbJr, 2021-05-23:
I ordered from jlc-pcb as well. They told me, that the file contains 2 designs. As far as I understood they meant that the copper traces of the two boards are not connected. to solve that, one could maybe change the pcb design, so that the test points for the display extend to the other board and connect the two. this way one could test the display before soldering on the flex cable adapters. maybe that would help with debugging the circuit. But the few bucks are a reasonable price to pay and are not really worth changing the pcb files I think. thank you for providing this guid and all the files!

killdrith, 2021-06-13:
I am also having them tell me that the minimum breakaway tab is too small, and I cannot modify the files in picad because the formats are outdated.

Wbrjr, they just mean that you selected 1 design instead of 2 on this menu:
[image]

luispadron, 2023-01-23:
Did ya'll ever fix this?