Gameboypi/SPW #55: USB Type C

aethrys, 2021-05-04:
Hi, was wondering if you (or anyone else) had suggestions on adding USB C to this project.

I'm already planning to use a type c variant of the TP4056 module, but I saw chatter on the link port, #7 , and I'd like to make use of it too. To make it simple, I've seen some breakout boards but it looks it'd be a bit of a pain to line up.

Any advice?

Eilermoon, 2021-06-01:
I am about to start this project and would also love ti figure out a type-C option for this! It would make things much easier not having to carry around an extra charger for it.

ercan1441, 2021-06-01:
I used the USB type C variant of the TP4056. Works fine here.

aethrys, 2021-07-29:
Good to know. Either of you wire up the link port with USB, too?

stanyertom, 2023-03-05:
I tried to wire the link port to the usb pads and then splice a link cable to usb but I think I melted the link port or something because it didn’t work.