Gameboypi/SPW #56: Blank White Screen Issue

Eilermoon, 2021-06-29:
Hi all! I'm curious if someone here can help troubleshoot the issue I'm having here. I've got most everything soldered up onto the PCB and screen, but all I'm getting is a blank white screen when connecting power to the Pi. I've checked all connections for continuity according to the 'Screen Troubleshooting' section and everything is soldered correctly as far as I can tell. The only issue I can see with soldered is possibly bridged connections on the board connected to the screen (The third picture). I know they're not clean, I'm still getting better at soldering, but it looks clean enough. 

Any suggestions? Is this possibly a software issue? I wrote the included RetroPi image to the SD using Win32DiskImager. Thanks for any help!

[image]
[image]
[image]
[image]

gambino37, 2021-06-29:
Get some flux and reflow your joints first.  Then always test for continuity on neighboring pins to each other as well. You can have continuity to the correct points but you also have to make sure you don't have continuity between points that shouldn't have it.

Eilermoon, 2021-06-29:
Quick update: So if you look at the first images I posted, I stupidly had the screen soldered upside-down. So contact #18 on the screen was soldered to contact #1 on the board! I fixed that last night and went over all of the contacts checking for continuity after removing some solder. 

I definitely fixed some bridging that was going on between the contacts on the board, but now I've got it to where only a few on the screen board are continuous. Contacts 1 and 8 are continuous to each other (both grounds) (blue in the image below), and 9, 10, the 10R resistor and the '3.3' pad are continuous (red in the image below). This seems correct to me as I don't see any bridging, but let me know if this is an issue! I've double and triple checked both of the ribbon connectors as well as the other board and everything else looks ok.

I've also hooked up the Pi via mini HDMI to a monitor and RetroPi is flashed and running correctly, so it's definitely not a software issue. In addition, none of the buttons were working when I held one on the 'connect controller' screen. Perhaps this is a clue as what the problem is? I'm at a bit of a loss on what to check next.
[image]

gambino37, 2021-06-30:
You still need flux. Those joints still look sketchy.

Eilermoon, 2021-06-30:
> You still need flux. Those joints still look sketchy.

Should've mentioned I'm picking some up tomorrow to clean these up, Hopefully that fixes the problem.

gambino37, 2021-06-30:
It definitely will make your joints much smoother. And flowed properly.

InterplanetaryGentlemann, 2021-06-30:
I just finished my own build for this and it was a pain in the rear to troubleshoot. Hopefully I can send in my updated version of the image to be added to this repo instead.

If after cleaning up the solder you still experience the issue, it is absolutely a software problem. The screen connects to the pi via its GPIO pins, which require different drivers than the HDMI output. The image provided in this repo has a display driver installed, but it did not work at all for me either and I ended up having to install a new driver as suggested by issue #42

Git for the drivers here, follow the installation instructions for the DILI9341 type of display. You can reference issue #42 for the cmake command that coincides with this hardware and the GPIO pins that need to be specified.
https://github.com/juj/fbcp-ili9341

If after install you still get a white screen, re-do your build directory for the drivers and add the -DUSE_DMA_TRANSFERS=OFF flag to the cmake. If it works after that, you still need to change/disable a setting that is colliding with the driver, then redo the install again without that flag.

A final thing to be aware of with this driver is that changing the base clock speed of the pi through setting 'core_freq=#' in your /boot/config.txt screws with something in the driver and causes your screen to show the video with a lot of pixelation/distortion.

Eilermoon, 2021-06-30:
> I just finished my own build for this and it was a pain in the rear to troubleshoot. Hopefully I can send in my updated version of the image to be added to this repo instead.
>
> If after cleaning up [...]

Thanks so much for this info! I was beginning to suspect going through the GPIO pins was part of the issue as it had no issue going through the mini HDMI port and the buttons weren't working either, meaning no communication was going through the pins. I'll try this right after work!!

gambino37, 2021-06-30:
If the buttons weren't working gpio still could be your issue.

Eilermoon, 2021-06-30:
> I just finished my own build for this and it was a pain in the rear to troubleshoot. Hopefully I can send in my updated version of the image to be added to this repo instead.
>
> If after cleaning up [...]

@InterplanetaryGentlemann just responded in #42, but do you have the image that worked for you? Not sure how I can connect to it to input commands.

jooogz, 2022-07-09:
> I just finished my own build for this and it was a pain in the rear to troubleshoot. Hopefully I can send in my updated version of the image to be added to this repo instead.
>
> If after cleaning up [...]

I know this is nearly a year ago, but is there any chance that you could share your modified image?

Eilermoon, 2023-04-15:
> I just finished my own build for this and it was a pain in the rear to troubleshoot. Hopefully I can send in my updated version of the image to be added to this repo instead.
>
> If after cleaning up [...]

Hey @InterplanetaryGentlemann I realize it's been 2 years but I just picked this project back up. Your suggestion worked great! Thanks so much, I figured better late than never.

closed 2023-04-15 by Eilermoon

Some1TGuy, 2024-02-05:
> Quick update: So if you look at the first images I posted, I stupidly had the screen soldered upside-down. So contact #18 on the screen was soldered to contact #1 on the board! I fixed that last [...]

I realize this is super old, but hopefully you see this.. I'm running into the exact same issues that you described with the white screen, so far all my gpio continuity testing is fine (not completely done yet) but I will try rebuilding the driver tonight. I'm also seeing the issue with the buttons not working when I'm connected through HDMI on the splash screen. I used the instructions to map GPIO from #60 

Any chance you can share your working image?