Gameboypi/SPW #57: Radxa Zero

aethrys, 2021-07-29:
Hey guys 👋 

I haven't purchased parts yet for my own build, and I'm a little glad I did wait because I recently learned of the Radxa Zero. Same form factor as the RPI0 with "updated" components.

I was curious about any opinions on it potentially being a "drop-in" replacement with this particular set of PCB and components. Immediate issues that come to mind is battery-life, and charging, but I wasn't quite certain about everything else.

https://forum.radxa.com/t/introduce-the-radxa-zero/6550

TranceLove, 2021-08-07:
Glad I'm not alone dreaming Radxa Zero to be in the heart of Gameboy Pi.

AFAIK Radxa Zero's pinouts are entirely different from RPi zero, almost certainly you will need a new PCB design.

Battery life would be a great challenge I think... emus requiring horsepower (that Radxa Zero is capable of than RPi Zero) would consume the battery within an hour.

I am no expert of this field, but I too wish someone could step up and fork this for Radxa Zero, and make it not just run Retropie (or EmuELEC for Amlogic) but TwisterOS as well to turn the little GBA into __a laptop in disguise__ ;)

(Although, you are expected to bring your USB hub and keyboard/mouse along.)

aethrys, 2021-08-31:
Yep. I haven't seen a pinout comparison yet, just a few spec sheets (1), but others have asked the same question (2)... The amlogic chip on this board is using a different pinout, despite having 40 pins (they're advertising that a lot, lol), and will not allow "drop-in" usage on existing pi-hats, etc. 

Since this project is using SPI for the display, and the Radxa Zero supports it (3)... I wonder if it's still doable. I really like how clean this build is, so having extra wires for bridging is putting me off though :/

I would need a pinout comparison to even attempt it, and I haven't dug all that deep for one.

I was planning to do two builds, as gifts, since the PCB supplier requires, like, 3 prints as a minimum. I think I'm going to go ahead and order parts for those two, and I may toy with the idea of doing a third if the Radxa Zero ever gains popularity. I'm hopeful, but I have doubts unless some more ready-to-go projects come along or a kind passer-by updates an existing designs (like this one!). 

(1) - https://sbcfinder.com/compare.php?sbconename=Radxa%20Zero&sbconetwoname=Raspberry%20Pi%20Zero%20W
(2) - https://forum.radxa.com/t/introduce-the-radxa-zero/6550/44
(3) - https://wiki.radxa.com/Zero/getting_started