Gameboypi/SPW #65: PSA This project is still good to go early 2023! zero & zero2

maarten682, 2023-03-12:
Ive been on the fence for quite a while about building a couple of these because of some of the problems mentioned in the issue tracker (pcb incompatibilities, zero2 issues, software problems) but decide to take the plunge anyways.

I can happily report that this project is still completely solid. No problem sending the gerber files to JLCPCB, other than the links on this github being slightly outdated i had no problem finding the correct parts on aliexpress (including a clone of the adafruit audio amplifier that works perfectly fine) and getting the screen and buttons working under the latest version of retropie on a zero2 were really not that difficult.

I used the following to get retropie 4.8 up and running;
https://github.com/juj/fbcp-ili9341 - screen drivers
https://learn.adafruit.com/retro-gaming-with-raspberry-pi/adding-controls-software - buttons
https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/raspberry-pi-usage - optional audio

I used triggerhappy to remap some key combos to be able to control things like volume ('amixer sset 'PCM' 10+') and safe shutdown.

If you are on the fence, dont. This is a very fun project. Battery life on a zero2 is not great on the small battery so keep that in mind.

Drippydh, 2023-03-15:
How did you got the battery working? I'm only able to power it on when it's connected with USB-C..

maarten682, 2023-03-19:
> How did you got the battery working? I'm only able to power it on when it's connected with USB-C..

It should work out of the box. Sounds like your Tp4056 board isnt soldered on correctly. Double check connection between battery terminals and the charge board two inner connections (B+ and B-).

Eilermoon, 2023-04-19:
Where the heck did you find a Zero 2W recently? Looking for a couple to use with new builds, but everywhere is out of stock!

Thanks for the Audio and button mapping tips 🙂

maarten682, 2023-04-19:
You cannot really buy them anymore unless you get stupid lucky or know an inside man, been like that for years now.

vantablack333, 2023-05-22:
> You cannot really buy them anymore unless you get stupid lucky or know an inside man, been like that for years now.

Why is this the case? The pandemic is over and there seems to be no chips deficit anymore

Eilermoon, 2023-05-26:
> Ive been on the fence for quite a while about building a couple of these because of some of the problems mentioned in the issue tracker (pcb incompatibilities, zero2 issues, software problems) but [...]

It looks like this audio install does not work for newer versions of Retropie. Anyone know other options?

```
pi@retropie:~ $ curl -sS https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/i2samp.sh | bash

--- Warning ---

The i2s amplifier installer
does not work on this version of Raspbian.
Check https://github.com//
for additional information and support
```

maarten682, 2023-05-27:
> It looks like this audio install does not work for newer versions of Retropie. Anyone know other options?

Yeh, just skip the fast install and scroll down to the detailed description.

CurlyAwesome, 2023-08-16:
Would you be so kind as to provide your RetroPie image? I'm pretty new to working with custom built linux emulators and can't for the life of me figure out why it's not working.

maarten682, 2023-08-16:
> Would you be so kind as to provide your RetroPie image? I'm pretty new to working with custom built linux emulators and can't for the life of me figure out why it's not working.

Like i wrote i just used the standard 4.8 image, afaik that is still the current version.

What exactly is 'not working', is your device not booting at all? Have you tried disconnecting the screen and battery and just powering the pi directly with a monitor attached to see what's happening?

CurlyAwesome, 2023-08-16:
I'm not good at working with linux and can't get the provided retropie image to emulate GBA smoothly, the default emulator crashes on startup and the other one in it runs poorly. I tried getting 4.8 to work but can't figure out how to install the extra needed plugins so that it uses the screen and such.

JosephBerman, 2023-11-13:
how does the PCB sit in the shell? do you think it is possible to create a shell mod to put a bigger battery in the cartridge slot. I want to build my own but I am worried about the battery life making the usability of the device trivial.

maarten682, 2023-11-15:
> how does the PCB sit in the shell? do you think it is possible to create a shell mod to put a bigger battery in the cartridge slot. I want to build my own but I am worried about the battery life making the usability of the device trivial.

The pi sits right in front of the cartridge slot so there is not a whole lot of space to work with, you could at best get a 3~4mm thick battery in there (a bit more if you remove the hdmi and usb ports from the pi). You will need to remove a lot of plastic to make it worth it including the cases original battery tray and most of the structural support so the bottom will become very weak and flimsy. You will also need a new solution for the shoulder button springs. Youll also need to make sure that your battery wont get punctured by any components if you push on the bottom of the device but if you do all of that you could probably get a fairly large battery in there (2~2,5Ah give or take).