Gameboypi/SPW #66: Charging board and buttons not working

Eilermoon, 2023-04-15:
Hi all! Hoping there are a few people following this.

I have everything wired up and should be ready to go. Powering the board via the Pi Zero's micro USB works correctly with the screen, but EmulatorStation does not detect any gamepad input so the buttons don't work. I've continuity tested all the buttons, which all were correct. 

I also can't seem to power the board via the USB-C port on my TP4056 Charging board - the LED on that board doesn't even light up.

Any feedback would be very helpful! I have a SP charger port too but don't have a charger to test it yet - I'll update when I can try that as well. Some pics attached that might help

[image]
[image]

Eilermoon, 2023-04-19:
Just adding an update: My board works now, but I'm not clear on what the issue was previously. I have since soldered on a SP Charger, which worked correctly and lit the LED on the TP4056 board. I also tried a different plug and USB-C cable, which also worked in the USB-C port! 

I've assembled everything and it's working great - the battery charges and is functional without being plugged in. I'll post an update and close this issue once I'm sure the issue was just the voltage input of the USB-C I was using previously.

maarten682, 2023-04-20:
Not sure if that was the case here for you but USB-PD sources do not work with these charging boards. You pretty much need good ol usb-a to c cables to make them go.

Eilermoon, 2023-04-20:
> Not sure if that was the case here for you but USB-PD sources do not work with these charging boards. You pretty much need good ol usb-a to c cables to make them go.

TIL what USB-PD is. But yes, I think you hit the nail. I was using a USB-C -> USB-C cable inbto my laptop, then from a wall socket and still didn't work. Must be the cables. Although I've tried a few USB-A -> USB-C from an outlet that didn't work as well, must've been too low Output. Thanks for the info!

closed 2023-04-20 by Eilermoon

maarten682, 2023-04-20:
Yup pretty much every usb-c port that outputs power needs to do a bit of handshaking before it starts sending out 5v.  These charging boards dont do any of that, they pretty much just rely on legacy usb that always provides 5v even if you stick a fork in there. Most (but as you found out not all) usb-a ports should be able to do that. Power can become an issue if you charge the battery and run your pi and screen all at full blast but sign of life should be possible from most sources (even ports on reasonably modern laptops), just your pi booting doesnt pull that much power.